astralien3000/aversive-- #67: Revise the whole build chain to properly make use of qmake

Elioty, 2015-06-23:
I looked a bit more into how qmake works. People usually use it through Qt Creator and I think it would be great to be able to work on/with Aversive++ within Qt Creator.

First, without taking Qt Creator into consideration; qmake, when generation makefiles, has a `-spec` option that allows the user to specify a directory containing a `qmake.conf` file. This file must contain everything required to build your project for a given target platform. So we could provide with Aversive++, our own spec files to build the lib for the different platform supported. The user would then just have to type something like `qmake -makefile -o Makefile -spec kits/atmega2560 aversive.pro` for example to generate the makefiles (not sure what is the `-o` option's default value though) and then just type `make`. Note that the spec directory is also automatically added to the `INCLUDEPATH` qmake variable and so we could put a CurrentPlatform.h file into each spec directory to give meta-data about the platform building for (like `typedef ADM2560 CurrentBoard; typedef AVR CurrentArchitecture;` with the proper includes before).
This prevents us to have a single makefile to compile for all the supported platforms at once but we could just write a simple shell script doing something like `qmake -makefile -o Makefile.$1 -spec kits/$1 aversive.pro` where `$1` is a command line parameter specifying the platform the user want to compile for. Then the user would execute `make -f Makefile.$1`. This way, we don't have conflicting Makefile file names and just need a simple shell script to execute them all if we still want to compile Aversive++ for all the supported platform at once.

Now within Qt Creator, people use compilation kit that are automatically detected/created upon Qt build environment installation. However, one can manually add additional kits and compilers. I have yet to investigate this a bit further to see if you can really use Qt Creator fully as some parameters would be totally irrelevant for an embedded use case and I hope there is a way to make Qt Creator ignore them in some way.

What do you think of completely switching to a .pro project?

astralien3000, 2015-06-23:
I am not sure that qmake will fit all our needs, but since it is already partially used in v14.11, it seems ok.
The other solutions would be CMake, SCons, Autotools, etc...
I don't know how to use SCons and Autotools, so I thins think it is not a really good idea to use those. About CMake, I have already tried it in the branch "modules", but I finally removed it, since it did not fit my needs in this branch. I think we would use qmake and Cmake the same way, but qmake is quite easier to use...
And I did not know the "spec" option in qmake, which seems interesting.
May be we should do a proof of concept (not too hard), and evaluate its maintainability.

Elioty, 2015-06-23:
Ok, after further investigation tonight, it seems that cmake integration within Qt Creator is good enough to be used (though one has to run cmake twice from the IDE when changing from one target to another) so we will integrate cmake soon into Aversive++ and probably wipe totally the dependency towards qmake (though Qt Core 5 is still required to compile Aversive++ Sasiae).

Elioty, 2015-10-30:
This will need some proper work when adapted to the project new directory tree but it will be done along the actual new implementation. So I close this issue.

closed 2015-10-30 by Elioty