astralien3000/aversive-- #78: Hardware Abstraction Layer (HAL)

astralien3000, 2015-10-04:
Here it is ! @Elioty and I are working on this for a few weeks.

The HAL is currently located in `/include/*/hardware` directory. The problem is that the interface provided for STM32 and AVR is not the same. The project is to have a new common interface, that will be available to the user.

Indeed, we concluded that it is not possible to provide generic devices and to enable the user to use every options of the hardware. So we decided to split the configuration of the hardware, and the use of the device. So that, the user will be able to use the full potential of his microcontroller, while being able the use the simple device interface.

The functionalities that are not available on an hardware are marked as deprecated, and may trigger a compilation error. As always, the error message should be clear.

The interface is almost done for GPIO, TIMER, and UART modules.

Elioty, 2015-10-15:
I reviewed your GPIO driver interface and I think some template methods are missing.

Indeed, for the methods which take two parameters, we have the full "dynamic" one taking just two C-like parameters and we have the full "static" one taking just two template parameters. But I think for every methods taking two parameters, we should also have one taking one template parameter (the pin number or the pin mask) and one C-like parameter (the Settings, the Mode, the OutputMode, the Alternate, etc).

I think it's at least very important for the pin value setter. We currently have:
```C++
static void setPinValue(bool value, u8 pinNumber);
template<bool VALUE, u8 PIN_NUMBER> static void setPinValue(void);
```
But a DigitalPin device will certainly have an interface (roughly) like that:
```C++
template<u8 PIN_NUMBER>
class DigitalPin {
public:
  setValue(bool value);
};
```

(I said roughly, I omitted a lot of things, I just kept the things important for the point I am making).

So here the pin number is templated but the value is dynamic, and we don't have a corresponding method in the driver interface. Which could be either fixed by a call to the fully dynamic driver method or by two call to the fully static driver method within an ```if then else```, but both aren't a good way of making this work in my eyes.

What do you think?

astralien3000, 2015-10-15:
Actually, I think the compiler can do a lot of optimizations (if the function is inline and a parameter can be evaluated at compilation, the code is exactly the same as the templated version).

The only difference between the templated version and the dynamic is the compile-time verifications the driver can do. Is it worth expanding as much the driver interface, which is already REALLY FAT ?

For example, the compile time verification for the Pin would be in the PIN_NUMBER, that we can do in an other function (may be the constructor) ?

Elioty, 2015-10-15:
Then maybe we shouldn't even bother about providing the fully templated methods but provide only the dynamic ones?

astralien3000, 2015-10-15:
As I said, the templated ones are able to do some compile-time checks, which is important, I think.
It would still be useful for the user when he uses directly the drivers.

The thing is that the templated and dynamic can have really differents implementations.
And even the half-templated functions can do some complie-time checks, but I fear that it would be too much for the devlopper to implement 4 times the same function. More over the 2 half-templated functions implementation will only be a mixed version of the 2 "pures" implementations.

astralien3000, 2015-10-15:
Maybe you can go and see the [ATMegaxx0_1 implementation](https://github.com/astralien3000/aversive--/tree/hal-atmegaxx0_1/hal/avr/mega/xx0_1), to have an application.

astralien3000, 2015-10-16:
Well, I found an other solution, that you may find better.

Elioty, 2015-10-16:
Oh yeah, good solution. However, we don't have the fully dynamic methods anymore :laughing: :sweat_smile: 
I agree it's a lot of methods (and work to do) but I think the more complete we are, the better.
So I would just add the fully dynamic methods in `DriverInterface`.

Another thing, how are drivers supposed to remove the deprecated attribute on enum values they support?

astralien3000, 2015-10-19:
Here it is, the 200 lines GPIO DriverInterface

To remove the deprecated attribute, the driver has to redefine the structure, which inherits from the interface, and redefine the accepted values.

Elioty, 2015-10-19:
"Wait a second..." I was getting into a long wall of text and then I saw your GPIO driver implementation for AtmegaXX0_1.

Ok so my point will be way easier to explain. Why is `struct Mode` defined within the concrete GPIO driver for AtmegaXX0_1 but is defined outside of the driver interface in the common files? I know all the other enums are unusable on AVR and their respective methods unavailable. But it makes the actual interface different across different drivers which I think could be very easily fixed just by defining the the different structures representing the enums in the driver interface. Then you keep doing the same way you did in your concrete driver. Do you see and agree on my point?

Same thing for the `using` which is defined in `HAL::Private::GPIO` and not in the driver interface.

One more argument to support my point, thing is that the way it currently is, the user cannot access everything through the driver interface, some things (which can simply be unsupported I agree, or not like the `using` declaration) are only accessible from `HAL::Private::WhateverModule`, which the final user (the one who uses the lib, not the one who implements drivers) should not even touch. And just in case, it's better for meta-programming because you can access everything you need (and even more) from the concrete driver interface you use. The problem currently (to give both the driver interface and the left-away "enum struct" or `using`) is that we cannot give namespaces as template parameters.

So I think I would (for example with the GPIO module) just move `struct HAL::Private::GPIO::DriverInterface` to `struct HAL::Private::GPIODriverInterface`, move everything which is left in `namespace HAL::Private::GPIO` (so the `using` and the "enum structs") directly into `struct HAL::Private::GPIODriverInterface` and then remove the now empty `namespace HAL::Private::GPIO`.

Anyway, I think you got it by now!

astralien3000, 2015-10-20:
Oh ! Yes, this is a problem I did not think of.
I think your solution is ok.
(I had to read your wall of text twice to understand it completely ^^)

Elioty, 2015-10-20:
Ok I checked your commits a bit and more particularly for the GPIO driver interface (so I may be missed some mistakes if there are any) and it looks good to me now. However, there are a few spelling and/or grammar mistakes in the documentation but I will fix them once your work is done, just before the merge into master.

Regarding your question, I actually don't know and I am a bit torn in two. It's semantically an interface so it deserves to be public. But on the other hand, the final user shouldn't use it directly (everything is tagged deprecated in it anyway) and use the driver of the concrete module in his CPU.

So I think I am ok with both solutions, the interfaces being in `::HAL::Private` or directly in `::HAL`.

astralien3000, 2015-10-22:
As we were discussing by mail, I think it would be better to replace "deprecated" secification from the interface (which is dirty, anyway), and replace it with just a protected or private interface. The error would not be at link-time, but at compile-time.

And also, maybe we should move the `pin_number` and `pin_mask` parameters to the begining of parameters list.

Elioty, 2015-10-22:
For the first point, isn't there any other attribute like deprecated which could throw an error rather than just a warning? If such an attribute exists, I would prefer it over making the methods private, cause semantically both making it private or tagged deprecated is not what we mean. In both case, I agree that the sooner the error is thrown up, the better.

For the second point, I really missed it!! (I would have seen it hopefully during my audit before accepting the PR). So yeah, I totally agree.

astralien3000, 2015-10-22:
I can try to do some research (again), but it was the only thing we have found.

I will do the second point tomorr... today. The same thing can be applied to timers OutputCompare

Elioty, 2015-10-22:
For the first point again, if you can't find any attribute that could throw an error, I would say make the methods private.

And by the way, we can also extend it to the enums. As Ryder stated in the email-chain, the user can access the enum values both from the struct attributes we declare (to add the deprecated attribute) and directly from the enum within the class. The problem with the second is that there is no warning issued. So I think we should make the enum declaration private. That shouldn't prevent the user to use the attributes we declare ourself in the struct. And same as for the methods, if you can't find any attribute to replace deprecated, we can just declare the attributes private and the driver developer will declare them again in the concrete struct as public.

astralien3000, 2015-10-23:
Well. Finally, I let the deprecated attribute.
BUT : I renamed the define DPRECATED into UNAVAILABLE.
And now the deprecated warning will display a message that explicitely says that the function the user is trying to use is not available on his microcontoller.

I think the interface declaration and the warning message are clearer, event if we use "deprecated" internally.

The private technic would have created problems that were not convenient for the divers developers.

Elioty, 2015-10-23:
What problems would the "private method" have caused to the driver developers?
And I still think it would be better to raise an error rather than a warning.

What about the = deleted ?

astralien3000, 2015-10-23:
The problem with your technic is that we must copy-paste the message at the end of each line... Which is not really cool.
Deprecated is beter for that, and if you want to generate an error, we can enable by default the flag -Werror in our makefiles.

Elioty, 2015-10-24:
As we discussed today, we will stick with the UNAVAILABLE macro define which uses the compiler deprecated attribute internally.

And for enum, we leave it like that for now.

Elioty, 2015-10-30:
Here is some documentation and proper tests made by Atmel:
http://www.atmel.com/images/doc8453.pdf

My point here is based on the first tip, page 4.
So simply, the problem is that the IO methods return or take as parameter 32-bit values. However, many MCU modules work only with 8 or 16 bits of data.
For example, AVR GPIO handle only 8 bits of data. But our interface only works with 32-bit values, which gives extra unnecessary work to the AVR MCUs (which are only 8-bit!) to handle these values.

A simple solution would then be to "templatize" the driver interface with the IO data type and use that type for the IO methods.

astralien3000, 2015-10-30:
TODO : Check if this is possible

```c++
template<typename T>
struct A {
  template<T VAL>
  struct B {
    ...
  };
};
```

Elioty, 2015-10-31:
Oh by the way, for meta-programming capabilities, we need a `using IO_Type = T;` (or another more appropriate name) in the templated driver interface so the user can easily use that type independently of the actual driver/hardware he is working with.

closed 2017-04-10 by astralien3000