astralien3000/aversive-- #83: Arduino-Syle HAL

astralien3000, 2015-10-21:
Hey !

I just had a discussion with one of my coworker, that was not really confident with C++, and that commented the HAL interface.
He said that it was way too scary to be used by someone that comes from the C way of thinking (even if he knows the C++ a little), and advised me to fit to the Arduino API.
I was a little "closed" to this idea, but the thing is that he is right. If we want to be used widely, it would be better to seem like something everyone understand !

Here is what i come with : 
 - We keep the C++ HAL interface.
 - We make a "C-style" warper that is proposed to the beginers.

BUT ! The "C-style warper" is not coded in C, but in preprocessor !
Actually, we make some macro aliases that just call the C++ templates interfaces.

So, we make just a simple overlayer that only the one that don't want to deal with C++ will use.

@Elioty  What is your feeling with this ?

Elioty, 2015-10-21:
I don't see any problem with proposing our own API (HAL + Device/Stream) and Arduino API (relying on HAL internally) side by side.

However, I am not sure the layer providing the Arduino API can be made in preprocessor relying on our fully templated interface simply because I am almost sure the Arduino API isn't made with preprocessor but with functions and methods. So we wouldn't actually fit the Arduino API if we make it that way because it wouldn't accept dynamic arguments at all. And for the GPIO, Arduino API just has only one "pin numbering space", I mean 1 to 54 (on the mega) and not 0 to 7 for each GPIO module like us so we would still need a big `switch` like it's actually already done in the official Arduino implementation I believe.

Moreover, two things:
- Is the Arduino API supposed to work for simulation purpose? Cause we decided that final user code for simulation would only rely on devices.
- With HDL and HAL, we made a big move towards the use of namespaces and even though I didn't talk about it yet, I thought we would/could/should use more of them for the rest of the API (a Math namespace like we already have, we could add Container, Geometry, Base, Device, etc). So would we provide a 100% compatible interface or should we just provide a very close API (with same function prototypes, no user's setup and loop function, and eventually provide (depending on previous question's answer) all this new API in an Arduino namespace)?

piroux, 2015-10-21:
> I am almost sure the Arduino API isn't made with preprocessor but with functions and methods. So we wouldn't actually fit the Arduino API if we make it that way because it wouldn't accept dynamic arguments at all.

*I heavily agree with that.*

Even if it seems to be a nice feature to have, it is also  a **bigbigtak* for the current state of the project while there are modules which have a higher priority, according to me.
But I think you should definitely keep it under the hood until the low level (in C++ ...) has been stabilized.

It is a bit weird to describe a C++ interface as low-level and a Cl interface as high-level ...

astralien3000, 2015-10-21:
Hehe, that's why I didn't called this issue "Arduino wrapper for HAL" ^^ !
We are not even expected to follow the Arduino API, but the point of my coworker was to avoid the use of templates and the "::" (namespace), basically...

We designed the HAL interface to be C-Style (not using objects), but maybe for some people it is not enough. The problem is that I am very confident with the template synthax, but I don't know at which point it is ugly for the others. And for the moment, what I see is that the people think the synthax is horrible.

To answer this, there is two solution : 
 - Either we say "I don't care", and the user is forced to use this interface.
 - Or we "sell our soul to the devil" an propose a fully (ugly) C warper

I a perfect worl, I would convince that the templates are the best thing they can learn ^^.
To me, the flat C style API is ugly, and I would not use this. But what can we say if a lot of people want it ?
I am just asking, I don't know if we will do it.

Elioty, 2015-10-21:
Gonna answer to your very last point first, ask as many things as you want, issues are there to discuss things.

As piroux said, I also think we should first focus on what we planned already. That's for sure.

And I think we should stick with the first "I don't care" solution (at least for quite some time). A fully C wrapper (so completely flat namespacing and no template) is totally ugly.
 - Namespacing allows to easily categorise things.
 - Templatings is for the meta-programming aspects (plus eventually to force the compiler to optimise things out) which is also really a great thing when you know how to use it properly, to let the user balance his performance/code size ratio himself.

Also, after a long email exchange with Rob'otter (here is a recap if you don't want to read all the emails haha), they pretty much don't get the namespacing and templating concepts we use. Not because they are incompetent (they are very good programmers) but because they simply (nearly) don't use C++ (nor advanced C++) at all and thus have no mental mechanisms to deal with it fluently. And that's a fact that it rather limits our audience to C++ advanced users (so not many electronics engineers).

However, we will hopefully have a rather extended documentation with some tutorials to get started easily (and to get how we use the namespacing and templating concepts in our API).

One last thing, we should provide proper evidences (mainly assembler code diff) to prove that our HAL is as efficient as the functionally-equivalent C code (with `avr.io` for example for AVR targets) because Rob'otter members were a bit suspicious about it. I am convinced we are/can be as efficient as the equivalent C code, we just have to write it all down with a proper and scientific process to make it a proof that we can easily redirect suspicious people to.

astralien3000, 2015-10-21:
Your last point has been transformed into issue #84 (with an EXELLENT title !)
So, for the moment, let's be the embassadors of real hardcore C++ style !

closed 2017-04-10 by astralien3000