astralien3000/aversive-- #86: A new language ?

astralien3000, 2015-10-22:
This feature will NOT be implemented before a long time. I just put it here to remember this idea that makes more and more sense to me.

Some of the constraints of C++ are becoming really inconvenient.
Maybe one day, it would be great to create a language that express exactly what we want, and does not force us to use tricky-borderline technics.

As creating a compiler from scratch is equal to suicide, the best option seems to create a frontend to LLVM. Some month ago, we rejected the use (and the support) of LLVM because it does not support AVR architecture. But I know that an AVR backend is in the process to be integrated to the trunk of LLVM !

[You can see that at the beginning of the README](https://github.com/avr-llvm/llvm)

Elioty, 2015-10-22:
Oh good news about AVR support for LLVM. It should become way more supported, fixed and regularly updated if it actually gets integrated into the official LLVM repo :D

And about making a new language, couldn't we just make extensions to C++11/14/17 ? (if you really wanna get into that rather BIG task, and to keep a similar syntax and "basic" functionalities)

Also, we should really message the C++ consortium, group, foundation (whatever is the jurisdictional form), well, the entity working on the C++ standards.

Maybe we could also list (in the first message of this issue) what extra features we would like to see implemented in the language?

astralien3000, 2015-10-25:
The problem with an extention of C++ is that the language is really hard to support entirely. Modifying a C++ frontend would be quite hard, I tink. But yes, it would be better than a new "from scratch" language.

So, let's list the features.

astralien3000, 2015-10-25:
Here we go for the 2 first features

## 1 - Mixing templates parameters and normal parameters

Quite weird, isn't it ?
Something like : 
```c++
void setMode(compil u8 ID, compil u8 PIN, Mode mode) { ... }
```
(The compile-time evaluated parameters are marked `compil`)
An other good keyword would be `static`, since it would not add a new keyword.
But for the examples, let's use `compil`, to be easyer to understand.

Would be equivalent to : 
```c++
template<u8 ID, u8 PIN>
void setMode(Mode mode) { ... }
```
The compile-time evaluated parameters are not forced to be at the begining of the parameter list.
The purpose is to hide the fact that we use templates A LOT to the users that don't like it ^^.
And it enables the second feature.

## 2 - Proposing an overloading method for compile-time evaluated parameters.

This enables to specialize any function that has normal parameters : 
```c++
void setMode(u8 ID, u8 PIN, Mode mode) { ... }
void setMode(compil u8 ID, compil u8 PIN, Mode mode) { ... }
```

That would extend the overloading feature  : 
```c++
void setMode(u8 ID, u8 PIN, Mode mode) { ... }
void setMode(u32 ID, u32 PIN, Mode mode) { ... }
```

Elioty, 2015-10-25:
## 3 - Allowing the developer of a templated class or function to constrain the interface of a given type parameter

Just like Java allows this with this syntax:
```Java
class MyClass<T extends InterfaceOrBaseClass> {
  // ...
}
```

It would also be really useful to be able to do it in C++ to force the final user to give something as parameter that follows a known interface (most probably defined by the original developer). It could allow the compiler to throw a clear error immediately when the template is used with a parameter that doesn't fulfill the constraint instead of a dark error when a method or an attribute of the given type cannot be found (because it doesn't follow the expected interface).

As for the syntax, C++ could use the same as Java with the extends keyword or use a colon (plus an optional access-specifier, not sure about the need of the virtual-specifier here) to looks more like the actual C++ syntax when you extend something.
```C++
template<class T : public InterfaceOrBaseClass>
class MyClass {
  // ...
};
```

Elioty, 2015-10-25:
I have an idea for the feature number one about the `compil` parameter flag.
Why not use `constexpr`? I think with the use it has already since C++11, it would make pretty clear it's a compile-time evaluated expression that the user has to use for the parameters which have that qualifier. And I tested to make sure, `constexpr` cannot be used in parameter list in C++14 so our addition wouldn't affect anything already in place.

astralien3000, 2015-10-26:
The idea is quite good. It seems better than `static`.

# 4 - Extending override and final to any method

The override specifier goes in pair with virtual. It is quite logical, since an interface is supposed to be virtual. But templates enable us to create interfaces without the cost of vtables.
Currently, the compiler will show an error if the parent's method is not virtual.
Here, the override specifier would mean that the method must be defined/declared in a parent class, wether it is virtual, or static, or anything.

Of course the final specifier has the same problem.
Here the final specifier would mean that no class that inherits the current class can define a method (virtual, static, ...) with the same name and signature.

astralien3000, 2015-10-30:
TODO : complete explanations
These are "bugs" of C++ specifications
5 - "Unable to specialize template outside of namespace"
6 - "Unable to specialize template in the same class, but partial specialization authorized"
7 - "static constexpr attributes declared in a class have to be defined, and takes RAM memory"

astralien3000, 2015-10-30:
# 8 - "function template partial specialization is not allowed"

astralien3000, 2015-10-30:
# 10 - "enable compil-time evaluated control structures"

astralien3000, 2015-10-31:
# 11 - Provide Mangling/Demangling function for symbols
The idea is to be able when we write asm statements to call C++ function or use C++ variables : 
```c++
asm("jmp " __mangle__(MyNamespace::MyClass::MyStaticMethod));
```
This will be quite hard to add, since C++ does not allow to use constexpr symbols in asm statements : 
```c++
constexpr const char* INSTRUCTION = "jmp __my_symbol";
...
asm(INSTRUCTION); // Error
```

astralien3000, 2015-10-31:
# 12 - Enable templates to get complex parameters
This include strings (const char*) and structures
```c++
struct A {
  constexpr int a;
  constexpr char* b;
};

constexpr A a = { 
  .a = 0,
  .b = "Test"
};

template<A PARAM>
struct B {
  const char* test(void) {
    if(PARAM.a == 0) {
      cout << PARAM.b << endl;
    }
  }
};
```
This one seems harder to implement...

astralien3000, 2015-10-31:
# 13 - Static reflection
I don't really care about the dynamic reflection, even if would also be great.
It may even be possible to do this as a library. But we need the compiler to provide some informations...
The aim is to be able to list a class' methods list. And get some information, that would be useful for templates. (This point is quite big, and I have no time to explain it now)

closed 2017-04-10 by astralien3000